M17-Project/TR-9 #4: AX5043 as RFIC?

radonyl, 2020-10-20:
since APRS(AFSK mod) and analog FM are listed in feature list, I'd recommend AX5043 as the RFIC which has **AFSK demod embedded** (vital for power saving by WOR mode) and also analog mode(even AM). check [this project](https://www.notblackmagic.com/projects/vuhf-radio/) for a proof of practise.

Btw, AX5043's sensitivity is higher -126 vs -122 with even lower price 1.2$ vs 2.4$ and half current 10ma vs 20ma.

sp5wwp, 2020-11-12:
What is the roll-off factor for the Raised Cosine filter (AMPLSHAPE=1)?
I can't find any info here:
https://www.onsemi.com/pub/Collateral/AND9347-D.PDF

sp5wwp, 2022-01-31:
AX5043 is a cool little chip. I'm testing its performance right now. M17 mode TX is possible.

sp5wwp, 2022-01-31:
See [this gist](https://gist.github.com/sp5wwp/c612fc8fe612e99c11f9b65cd50bca3c).

ng91, 2022-01-31:
It's nice chip because it has many modulations that other chips lack of.

f4grx, 2022-02-01:
Definitely!

@sp5wwp thats great! BTW I suggest avoiding floating point in embedded code :)