M17-Project/TR-9 #6: Dual band version

Supermagnum, 2020-11-09:
144–146 Mhz
432–438 Mhz 
20W RF on both bands.
Digital Voice (M17 Codec)
FM Voice
APRS via 1200 bps Packet (AX.25)
Short text messaging (SMS-like)
Optional Wi-Fi and GPS module ( UBLOX NEO-M8N )
Touchscreen with adapter that fits in a cars 2x DIN (180x100mm) hole,or suction cup for windshield.
Compatibility with Navit, relevant patch request: https://github.com/navit-gps/navit/issues/982

If going the DIN2 route:
Optional software functions: Media player (MP3 and common formats from USB key) Audio media level control.
4x50W sound output channels. +One sound input channel ( microphone ).
Bluetooth.
RTL2832U based receiver for DVB-T,Wide commercial FM,DAB+,Traffic Message Channel

iddq, 2020-11-10:
HW AFSK+DTMF en/decoder

kc1awv, 2020-11-10:
I think we're getting beyond the scope of a handheld radio here. This is better suited for a separate project, since this looks like an in-dash radio capable of transmitting on amateur radio bands.

This may also be a very niche product, as I don't know of any manufacturer that offers a double-DIN transceiver.

Closing due to out-of-scope for the TR-9.

closed 2020-11-10 by kc1awv

Supermagnum, 2020-11-10:
> I think we're getting beyond the scope of a handheld radio here. This is better suited for a separate project, since this looks like an in-dash radio capable of transmitting on amateur radio bands. [...]

Yes, but at least I would like a transceiver that has a remote head that can do navigation and display APRS stations close by.  There is no transceivers on the market who is capable of that. And, the APRS frequency here is  144.800 MHz

kc1awv, 2020-11-10:
> Yes, but at least I would like a transceiver that has a remote head that can do navigation and display APRS stations close by. There is no transceivers on the market who is capable of that. And, the APRS frequency here is 144.800 MHz. Also, our biggest repeater net uses CTCSS.

I understand what you are requesting, but the TR-9 is a handheld transceiver, not a mobile radio. For that reason, this is out of scope for the TR-9 project.

When a mobile radio is being developed, we will announce that via Twitter and our main chat channels.