M17-Project/TR-9 #14: HMI: Select the correct strap options for boot mode on the STM32

tkuester, 2020-11-12:
Per `ERRATA.txt`, pins 31 and 32 on U1 (STM32L011) should be shorted together to select the correct boot state. Tentatively, I've labeled this "bug", although I'm not sure what the implications of it are yet.

I suspect the best way to do this is with a pull down resistor. That is a little less "permanent" of a change, and permits easier bodge-wiring at a later date. Perhaps we can reuse a 47k 0603 -- or alternatively, kicad's "Jumper" if we want to save on soldering another part.

I took a brief look at the [datasheet, section 3.4.4](https://www.st.com/resource/en/datasheet/stm32l011d4.pdf). It indicates that there are several pins which control boot mode, can anyone confirm this needs to take place?

sp5wwp, 2020-11-12:
We probably wont have to leave that pin floating, so a pulldown or even a hard-tie to GND is OK.

tkuester, 2020-11-14:
Doing a bit more reading on the STM32 boot modes. I now understand that `PB9` is `BOOT0`, and `nBOOT{0,1}` and `nBOOT_SEL` are configuration bits.

I had to push some traces around a little to get the resistor to fit, but I don't think this is worth pushing yet -- I still don't understand enough about the STM32 to consider this ready. Is there any reason we would want `BOOT0` to be logic high? Or would this be a recovery scenario, and a bodge wire is tolerable?

I'm going to make a separate issue with a documentation request for compiling and flashing the firmware -- I think this would help better understand what the board needs!

sp5wwp, 2020-11-14:
There's no reason to put logic high on that pin.

Since it can be fixed with a solder blob, I'd leave this as is, mention this in ERRATA.txt and move on.