M17-Project/TR-9 #7: Unify SWD header on TR-9 with HMI

tkuester, 2020-11-12:
As of 9ac1f05, the HMI board and the TR-9 share a different header layout for the SWD interface. Per a conversation with @sp5wwp, it would be nice to unify the headers to a common standard.

[image]

[image]

Some considerations I think are important:

 * Having the same ordering, and number of pins. If pin 1 (square pad) is GND on one board, it should be the same on the other
 * Having a common interface with common STM32 programmers. (Regrettably, I am new to the STM32 ecosystem, I'm not sure what's common.) SP5WWP indicated that the 1.27mm pin headers are frustrating to work with
 * A silk screen indicating to the user which pin does what
 * Consideration for accessibility. It would be nice to be able to access the headers without having to entirely disassemble the module!

sp5wwp, 2020-11-14:
- Pin 1 can be GND on both connectors. Both should have the same ordering.
- SWD uses SWCLK, SWDIO, /RST (and GND) signals. I'm not sure if there's any "common connector".
- Info on the silk screen is OK if one can manage to put it there :-)
- We can use anything as the connector. It's worth noticing that HMI board won't receive frequent firmware updates.

tkuester, 2020-11-15:
I have a proposal for the SWD connector.

Programmers like [this](https://www.amazon.com/HiLetgo-Emulator-Downloader-Programmer-STM32F103C8T6/dp/B07SQV6VLZ/ref=sr_1_7?dchild=1&keywords=swd+programmer&qid=1605455717&sr=8-7) are relatively cheap and easy to come by. (~$6 USD) Do you think we could use the header for this?

Alternatively, I have an STM32 Nucleo-64 development board on my desk, and the pin order is:

1 - VDD_TARGET
2 - SWCLK
3 - GND
4 - SWDIO
5 - nRST

I'm still new to the STM32 platform, I'm just looking for "standards" that are pre-existing.

sp5wwp, 2020-11-15:
We can use that mapping. Also: it is not necessary to use VDD_TARGET line.