M17-Project/TR-9 #24: Estimated cost and time to build?

kj7rrv, 2021-12-10:
Hello, 
I'm interested in building one or two of these. About how much does it cost (in USD) to build one, and how long does it take?
73, KJ7RRV

n2xdd, 2021-12-10:
Hey Sam , thanks for checking us out.   Right now any work on the tr9 is temporarily on hold, we have some work to do on the RF  side of it.  But in the meantime come join us on our discord server and see what other ways you can get up and running with m17.  
Links to discord are on our website . 

73 Ed N2XDD

kj7rrv, 2021-12-11:
Okay, thank you! It looks like there are also IRC and Matrix channels. Are they still bridged to discord? I would rather use a FOSS platform.

sp5wwp, 2022-01-16:
Matrix is linked. I'm not sure about IRC.